contiki-ng/contiki-ng #2507: Release 4.9 - task list before release

joakimeriksson, 2023-07-10:
# Preparations for Release 4.9
For a release, some additional testing beyond the regular automatic tests needs to be done, mainly to validate functionality on the different platforms. This extended testing phase plays a crucial role in ensuring that the software performs optimally across a variety of devices that Contiki-NG supports.

For each platform test at least the following examples:
- hello-world
- rpl-udp (the most basic communication)
- leds-example
- 6tisch/simple-node - TSCH on relevant platforms

## Documentation
While testing platform - check documentation consistency
- [x] #2315

## Platforms - arch/platforms
- [ ] zoul (Firefly or Remote)
- [ ] cc2538dk
- [ ] cc26x0-cc13x0
- [ ] simplelink
- [ ] nrf52840
- [ ] nrf
- [ ] Gecko / EFR32
- [ ] native (without communication)
- [ ] Cooja (just to test that it works outside the test framework)

joakimeriksson, 2023-07-10:
I am currently testing EFR32.

arurke, 2023-07-10:
I can test simplelink and cc26x0-cc13x0 (using only sensortag/cc2650 though). Will try to get it done tomorrow.

Opened one doc. PR on the release-branch: #2512

pjonsson, 2023-07-11:
The examples seem to work in Cooja running in the docker image.

The printout for version looks strange though: `[INFO: Main      ] Starting Contiki-NG-develop/v4.9`

joakimeriksson, 2023-07-11:
@pjonsson - Great - thanks for testing - The version is probably ok for now - as the tag is not the final yet - but the develop/v4.9 tag.

joakimeriksson, 2023-07-11:
I will test nrf52840 now.

pjonsson, 2023-07-11:
The nrf bullet should be split into sub-bullets, nrf52840 and nrf5340 are quite different since there isn't any network support with nrf5340.

The hello-world and leds examples work on nrf5340.

joakimeriksson, 2023-07-11:
> I can test simplelink and cc26x0-cc13x0 (using only sensortag/cc2650 though). Will try to get it done tomorrow.
> 
> Opened one doc. PR on the release-branch: #2512

I have two cc13xx - I can do a hello-world, leds and rpl-udp on them if you do the 26xx series.

joakimeriksson, 2023-07-11:
One thing that I lack on the startup messages is the board and target names - it would be really good to have that as it is guesswork now to understand what sits there (when having bunches of connected devices).

arurke, 2023-07-11:
Tested all the listed examples on simplelink and cc26x0-cc13x0, using a sensortag/cc2650. Also tested TSCH interop between the two.

arurke, 2023-07-12:
Last minute release note addition: https://github.com/contiki-ng/contiki-ng/pull/2520

nfi, 2023-07-12:
> I don't have heaps of devices, but doesn't `make -f Makefile.include motelist-all` give a reasonable output for that?

Unfortunately, only with very few platforms. For example, this is how output can look like from two different nRF boards, both built with platform nrf:

/dev/tty.usbmodemFEF45684353F1  FEF45684353F  0x1915  0x521F  Open DFU Bootloader  Nordic Semiconductor
/dev/tty.usbmodem0009601076211  0x393A245B  0x1366  0x1055  J-Link   SEGGER

joakimeriksson, 2023-07-12:
Thank you all for the release efforts!

closed 2023-07-12 by joakimeriksson