contiki-ng/contiki-ng #2566: printf of double and Contiki logging does not work in Cooja

rahav, 2023-07-28:
All of the following print statements would not print the number in Contiki 4.9 
```
printf("Double f = %f\n", 3.1);
printf("Double e = %e\n", 3.1);
printf("Double g = %e\n", 3.1);
LOG_INFO("Double f = %f\n", 7.1);
// and so on
```

pjonsson, 2023-07-29:
This bug was present in 4.8 as well.

The floating point printing part (`case CONV_FLOAT:`) of dbg-io is apparently missing. This was missing from commit 195c23aaa4, and has never been added or bug reported.

The reason Cooja was switched to dbg-io in commit f45a8c53274f was issues on Ubuntu 19 systems. This predates the 64-bit support in Cooja, and everything else that has changed since then (JNI -> FFI switch, using `-Wl,--wrap` on Linux, etc).

It's possible Cooja could stop using dbg-io, and cooja-log.c could use the regular functions in libc. This isn't a 30 minute fix though (and it needs testing on OSX too), and I don't see a simple workaround right now either.

rahav, 2023-07-29:
Thank you for the context. How do people use Cooja then? In my case, about 90% of the LOG_INFO, LOG_ERR, ... output doubles (for example probabilities between 0 and 1). So the entire data processing pipeline is now broken. I would like to upgrade from Contiki 4.7 but will wait to hear from you first if I should avoid it. Is the problem only in the Mote output window and I can output to a file instead? 

If there's no hope, how do I download the latest Contiki 4.7 container? I see some discussion [here](https://docs.contiki-ng.org/en/develop/doc/getting-started/Docker.html#using-a-different-docker-image) but how do I find out what 4.7 containers are available? What is their hash tag?

PR #2567 (2023-08-25): cooja: fix printf for floats
There is no support for printing floats
in dbg-io, so stop using that.

The issues that f45a8c5327 tried to
avoid are not present on my computer,
which could be explained by 64-bit support
in Cooja, FFI instead of JNI, or using
-Wl,--wrap for wrapping the printf-family
of functions.

Fixes #2566