contiki-ng/contiki-ng #2593: CC1352p-2 and cc1352r1 are not booting

jonathanns, 2023-08-10:
I tested from Contiki-NG version 4.3 to 4.9 and only 4.3 is booting the hello world example. Am i missing something? All version works with my CC2650, but not with my cc13x2 targets.

pjonsson, 2023-08-10:
I cannot speak for what happened 4 years ago, but I know for certain that the recent releases have been tested on (some) hardware, and the hello world example is one of the things tested. I would guess the earlier releases were tested in a similar way too.

Narrowing down the regression to after 4.3 is an excellent start though, but there are still 294 commits between 4.3 and 4.4. Since you have the hardware available, does 4.4 boot if you revert the commits in #978 and do a `git submodule update` after? If you want to avoid reverting things, you can do `git checkout <commit-sha>` for where you want to be.

pjonsson, 2023-08-10:
Ok, that is great progress. I'm guessing it's something with SDK 4.10.02, but I'm not sure how to pinpoint if further in an easy fashion. If you have time, it might be worth trying with 4.10.00.10 and 4.10.00.10 from https://www.ti.com/tool/download/SIMPLELINK-CC13X0-SDK/4.10.02.04, but that could just as well fail to build because of API changes from 3.60.01.

As far as I know, the people who know this hardware are on vacation for a couple of more weeks.

jonathanns, 2023-08-10:
That's unfortunate. I have been searching and this SDK only covers the CC13x0 devices .  So i guess someone combined the cc26x0 and cc13x2-cc26x2 SDKs into this one. Is it correct? Then from what SDK version the cc13x2 driver came from?
Lastly, The problem might be only on the SDK or also on the API?  Do you have any guesses where to start?

pjonsson, 2023-08-11:
Sorry, I searched for the SDK version but probably ended up giving you the wrong link. It might be https://www.ti.com/tool/download/SIMPLELINK-CC13X2-26X2-SDK/4.10.00.78, the release date is closer to the PR, but the version number isn't an exact match.

What I meant with the API changing is that you can try using a different SDK version. If you get a build problem, that is probably because some API changed compared to the default SDK.

The PR that updates the SDK (https://github.com/contiki-ng/coresdk_cc13xx_cc26xx/pull/2) is +29,421 lines and -201,145 lines, so we're looking for a needle in a haystack.

Out of curiosity, does commit 27516f3 (which is in the middle of #978) still boot for you?

jonathanns, 2023-08-11:
No problem, i understand. The SDK version is not quite clear. I have been reviewing the changes and really is a lot to see. It would take a long time to identify the problem.

And what i meant by mentioning the API is that the problem could also be there and not exactly on the SDK (which was also modified/adapted). Also, changing the SDK causes a lot of build problems.

The commit 27516f3ff2f545a2dc9ee5f01fa07571fb9763c4 only builds with the latest SDK on the simplelink submodule (done in the subsequent commit). The previous commits do not compile with both old and current core SDKs.

Is there  a log of which hardwares were tested after this release?

pjonsson, 2023-08-12:
I'm not suggesting you review 30 000 lines of code to find the issue, it will be far simpler to install some other version of the SDK and see if your device starts booting again.

For the 4.9 release, parts of the release work are noted in #2507, and this [comment](https://github.com/contiki-ng/contiki-ng/issues/2507#issuecomment-1630691456) mentions testing on the cc13xx devices.