contiki-ng/contiki-ng #2638: CC2538dk not transmitting

NestorESA, 2023-08-23:
I am creating a network (root and node) with TI CC2538dk, The nodes are supposed to use TSCH to create the network, i followed all the instructions from the tutorial in the website about CC2538dk. I flashed the software developed with Contiki-NG into the boards, they connect to the tunnel but they dont transmit anything. I am not sure if I am missing a command in the project files or any other step.

pjonsson, 2023-08-23:
Which version of Contiki-NG are you using?

NestorESA, 2023-08-23:
I believe I have the v4.6

NestorESA, 2023-08-28:
Do i  need to add the functions below to the C code to make the boards work?
platform_init_stage_one()
platform_init_stage_two()
platform_init_stage_tree()

kkrentz, 2023-09-08:
Depending on the ambient noise, you may need to increase the `CC2538_RF_CONF_CCA_THRES`.

NestorESA, 2023-09-08:
> Depending on the ambient noise, you may need to increase the `CC2538_RF_CONF_CCA_THRES`.

what does this mean?
still didnt manage to make the boards work as root and node in a TSCH network

kkrentz, 2023-09-08:
Sorry, the CCA threshold used to be a problem, but meanwhile TSCH seems to disable CCAs anyway by default.

NestorESA, 2023-09-19:
Does anyone have a solution for this?
the boards work with CSMA but not with TSCH, i dont even get to log into the board when I use TSCH