contiki-ng/contiki-ng #2659: Openmote-b bme280 sensors board using

geonugang, 2023-09-06:
# I created a file to test the bme-280 sensor in openmote-b by referring to the zoul bme-280 test file and Makefile supported by contiki-ng. However, when attempting to compile, an error occurred: no rule to make target 'sensor-test.upload'.

## The Makefile was created by referring to hello-world's Makefile and zoul's Makefile. However, an error occurs during the compilation process where the target cannot be specified, and I would like a solution.In addition, the openmote-b bme-280 sensor board test file was created by referring to the bme-280 test examples of hello-world and zoul.

[https://github.com/geonugang/Openmote-B.git](url)

pjonsson, 2023-09-06:
The project is called `sensor-test2`, not `sensor-test`.

I'm guessing it's not related to the problem you're seeing, but you don't want sensor-test2.c in CONTIKI_TARGET_SOURCEFILES.

If bme280 is a module, adding that to MODULES will compile the right files, no need to add it to CONTIKI_TARGET_SOURCEFILES.

I think you want to include Makefile.include last, not in the middle.

9morano, 2023-09-06:
I agree with @pjonsson. The Makefile should look something like:

`
CONTIKI_PROJECT = hello-world
all: $(CONTIKI_PROJECT)

CONTIKI = ../..

include $(CONTIKI)/Makefile.dir-variables
MODULES += $(CONTIKI_NG_DRIVERS_SENSOR_DIR)/bme280

include $(CONTIKI)/Makefile.include
`
Hope this helps :)

geonugang, 2023-09-07:
it works thanks!
I couldn't find 'bme280-arch.c', so I searched for it separately and added it, and it worked fine.

closed 2023-09-14 by geonugang