contiki-ng/contiki-ng #2669: Gitter is down

sungminlee114, 2023-09-10:
Hi all,

I'd like to ask few questions via Gitter, but found out that [gitter](https://gitter.im/contiki-ng) in the readme link is down with 404.
Is this intentional? If so, which channel will be appropriate for simple questions?

Thanks.

joakimeriksson, 2023-09-16:
Ok. I would suggest putting the questions here on discussions as we try to get most of them collected here!

We will update the readme.

marek22k, 2024-01-03:
Maybe join [#contiki-ng_Developers:gitter.im](https://matrix.to/#/#contiki-ng_Developers:gitter.im) or [#contiki-ng:gitter.im](https://matrix.to/#/#contiki-ng:gitter.im) directly?

micw, 2024-06-14:
It's https://matrix.to/#/#contiki-ng_Developers:gitter.im the second does not exist. The docs should be updated accordingly.