contiki-ng/contiki-ng #2823: linker error when compile hello-world for nrf52840dk

dingiso, 2023-12-26:
Dear Contiki-NG maintainers, 

I have compiled the hello-world example on nrf52840dk. Mutiple linker errors occured:

```
/data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-closer.o): in function `_close_r':
closer.c:(.text._close_r+0xc): warning: _close is not implemented and will always fail
/data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-fstatr.o): in function `_fstat_r':
fstatr.c:(.text._fstat_r+0xe): warning: _fstat is not implemented and will always fail
/data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-signalr.o): in function `_getpid_r':
signalr.c:(.text._getpid_r+0x0): warning: _getpid is not implemented and will always fail
/data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-isattyr.o): in function `_isatty_r':
isattyr.c:(.text._isatty_r+0xc): warning: _isatty is not implemented and will always fail
/data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-signalr.o): in function `_kill_r':
signalr.c:(.text._kill_r+0xe): warning: _kill is not implemented and will always fail
/data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-lseekr.o): in function `_lseek_r':
lseekr.c:(.text._lseek_r+0x10): warning: _lseek is not implemented and will always fail
/data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-readr.o): in function `_read_r':
readr.c:(.text._read_r+0x10): warning: _read is not implemented and will always fail
/data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /data/.local/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-writer.o): in function `_write_r':
writer.c:(.text._write_r+0x10): warning: _write is not implemented and will always fail
collect2: error: ld returned 1 exit status
make: *** [../../Makefile.include:654: build/nrf52840/dk/hello-world.nrf52840] Error 1
```
I have also test to compile the same example on cc2538dk. There are no errors output. I am not sure where these functions are used and why the things not work on nrf52840dk.

The compiler and contiki-ng version of me:

arm-none-eabi-gcc: `gcc version 13.2.1 20231009 (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7))`
contiki-ng: 
```
commit 658c2267951c4920a4a3c2575fb0865cf2e59416 (HEAD -> develop, origin/develop, origin/HEAD)
Merge: c45e3b945 392222497
Author: Niclas Finne <niclas.finne@ri.se>
Date:   Fri Dec 22 16:33:21 2023 +0100
```

Any Suggestions would help 

Best Regards,
Dingisoul

pjonsson, 2023-12-26:
It's a known issue, the link error started happening with GCC 12 from ARM because of some changes in their libc.

The `contiker/contiki-ng` docker image uses 10.3-2021.10, and I think Contiki-NG also works with GCC 11 from ARM.

dingiso, 2023-12-27:
Thanks pjonsson, 

I will try to use the docker image and reponse if all things go well soon.

Best Regards,
Dingisoul

marek22k, 2024-01-03:
The error also appears for me (I use https://archlinux.org/packages/extra/any/arm-none-eabi-newlib/):
```
$ make TARGET=nrf52840
which: no ccache in (/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
which: no nrfjprog in (/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
which: no nrfjprog in (/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
which: no rlwrap in (/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
  CC        ../../arch/platform/nrf52840/dk/board-leds.c
  CC        ../../arch/platform/nrf52840/dk/board-buttons.c
  CC        ../../arch/platform/nrf52840/dk/nrf52840dk-sensors.c
  CC        ../../arch/platform/nrf52840/./platform.c
  CC        ../../arch/platform/nrf52840/common/temperature-sensor.c
  CC        ../../arch/cpu/nrf52840/./dbg.c
  CC        ../../arch/cpu/nrf52840/./clock.c
  CC        ../../arch/cpu/nrf52840/./rtimer-arch.c
  CC        ../../arch/cpu/nrf52840/dev/uart0.c
  CC        ../../arch/cpu/nrf52840/dev/watchdog.c
  CC        ../../arch/cpu/nrf52840/./int-master.c
  CC        ../../arch/cpu/nrf52840/./slip-arch.c
  CC        ../../arch/cpu/nrf52840/dev/gpio-hal-arch.c
  CC        ../../arch/cpu/nrf52840/rf/nrf52840-ieee.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/components/boards/boards.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/components/libraries/util/app_error.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/components/libraries/util/app_error_handler_gcc.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/components/libraries/util/app_error_weak.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/components/libraries/util/nrf_assert.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/components/libraries/util/app_util_platform.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/components/libraries/queue/nrf_queue.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/components/libraries/strerror/nrf_strerror.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/modules/nrfx/drivers/src/nrfx_clock.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/modules/nrfx/drivers/src/nrfx_rtc.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/modules/nrfx/drivers/src/nrfx_gpiote.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/modules/nrfx/drivers/src/nrfx_rng.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/modules/nrfx/drivers/src/nrfx_wdt.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/modules/nrfx/drivers/src/nrfx_power.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/modules/nrfx/mdk/system_nrf52840.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/integration/nrfx/legacy/nrf_drv_clock.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/integration/nrfx/legacy/nrf_drv_rng.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/integration/nrfx/legacy/nrf_drv_power.c
  CC        ../../arch/cpu/nrf52840/lib/nrf52-sdk/components/libraries/fifo/app_fifo.c
  AS        ../../arch/cpu/nrf52840/lib/nrf52-sdk/modules/nrfx/mdk/gcc_startup_nrf52840.S
  CC        ../../os/dev/button-hal.c
  CC        ../../os/dev/gpio-hal.c
  CC        ../../os/dev/leds.c
  CC        ../../os/dev/nullradio.c
  CC        ../../os/dev/serial-line.c
  CC        ../../os/dev/slip.c
  CC        ../../os/dev/spi.c
  CC        ../../os/lib/aes-128.c
  CC        ../../os/lib/assert.c
  CC        ../../os/lib/ccm-star.c
  CC        ../../os/lib/circular-list.c
  CC        ../../os/lib/crc16.c
  CC        ../../os/lib/csprng.c
  CC        ../../os/lib/dbl-circ-list.c
  CC        ../../os/lib/dbl-list.c
  CC        ../../os/lib/heapmem.c
  CC        ../../os/lib/hexconv.c
  CC        ../../os/lib/ifft.c
  CC        ../../os/lib/iq-seeder.c
  CC        ../../os/lib/list.c
  CC        ../../os/lib/memb.c
[... cut ...]
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(libc_a-writer.o): in function `_write_r':
/build/arm-none-eabi-newlib/src/build-nano/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/writer.c:49:(.text._write_r+0x10): warning: _write is not implemented and will always fail
collect2: error: ld returned 1 exit status
make: *** [../../Makefile.include:654: build/nrf52840/dk/hello-world.nrf52840] Error 1
```

pjonsson, 2024-01-03:
@marek22k yes, it is a known issue. I don't know how the arch packaging system works, but for what you download from ARM directly: use a GCC older than 12 and things should link for you.

marek22k, 2024-01-03:
Thank you for your answer. Arch Linux is a rolling release and always uses the latest version, but even Debian Stable uses gcc 12. Is there a workaround to compile it with gcc 12/13? Docker does not work for me because I use nftables as a firewall and Docker only supports the outdated iptables.

dingiso, 2024-01-03:
I think you can download a compressed file of arm-none-eabi-gcc and unzip it into one place such as ~/.local/bin and add it to your PATH or you may choose to use something like miniconda

marek22k, 2024-01-09:
It does not work with GCC 11 either (followed https://docs.contiki-ng.org/en/develop/doc/getting-started/Toolchain-installation-on-Linux.html#install-arm-compiler and used GCC 11).
```
$ make TARGET=nrf52840 BOARD=dongle clean
which: no nrfjprog in (/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
which: no nrfjprog in (/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
which: no rlwrap in (/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
$ export PATH="/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:$PATH"
$ make TARGET=nrf52840 BOARD=dongle
which: no nrfjprog in (/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
which: no nrfjprog in (/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
which: no rlwrap in (/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/:/home/marek/.rvm/gems/ruby-3.2.2/bin:/home/marek/.rvm/gems/ruby-3.2.2@global/bin:/home/marek/.rvm/rubies/ruby-3.2.2/bin:/home/marek/perl5/bin:/home/marek/.local/bin:/home/marek/.nvm/versions/node/v20.10.0/bin:/home/marek/.pyenv/shims:/home/marek/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/marek/.rvm/bin:/home/marek/.rvm/bin:/home/marek/go/bin)
  MKDIR     build/nrf52840/dongle/obj/.deps
  CC        ../../arch/platform/nrf52840/dongle/board-leds.c
  CC        ../../arch/platform/nrf52840/dongle/board-buttons.c
  CC        ../../arch/platform/nrf52840/dongle/nrf52840dongle-sensors.c
  CC        ../../arch/platform/nrf52840/./platform.c
  CC        ../../arch/platform/nrf52840/common/temperature-sensor.c
  CC        ../../arch/cpu/nrf52840/./dbg.c
  CC        ../../arch/cpu/nrf52840/./clock.c
  CC        ../../arch/cpu/nrf52840/./rtimer-arch.c
  CC        ../../arch/cpu/nrf52840/dev/uart0.c
  CC        ../../arch/cpu/nrf52840/dev/watchdog.c
[... cut ...] disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-gnu-as --with-gnu-ld --with-sysroot=/data/jenkins/workspace/GNU-toolchain/arm-11/build-arm-none-eabi/install/arm-none-eabi --with-multilib-list=aprofile,rmprofile --with-pkgversion='Arm GNU Toolchain 11.3.Rel1' --with-bugurl=https://bugs.linaro.org/
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 11.3.1 20220712 (Arm GNU Toolchain 11.3.Rel1) 
```