contiki-ng/contiki-ng #2834: Cooja OutOfMemory Error

ArshdeepSingh18, 2024-01-07:
Dear Contiki-NG Developers,

I've encountered an issue while attempting to create a larger network in Cooja. After a while, it consistently throws an OutOfMemory error in Java. Is there a possibility of a memory leak within Cooja causing this problem?

```sh
Exception in thread "sim" java.lang.OutOfMemoryError: Java heap space
        at java.base/java.util.Arrays.copyOf(Arrays.java:3482)
        at java.base/java.util.ArrayList.grow(ArrayList.java:237)
        at java.base/java.util.ArrayList.grow(ArrayList.java:244)
        at java.base/java.util.ArrayList.add(ArrayList.java:483)
        at java.base/java.util.ArrayList.add(ArrayList.java:496)
        at org.contikios.cooja.plugins.TimeLine$MoteEvents.addRadioRXTX(TimeLine.java:2051)
        at org.contikios.cooja.plugins.TimeLine$14.accept(TimeLine.java:929)
        at org.contikios.cooja.plugins.TimeLine$14.accept(TimeLine.java:864)
        at org.contikios.cooja.util.EventTriggers.trigger(EventTriggers.java:111)
        at org.contikios.cooja.contikimote.interfaces.ContikiRadio.signalReceptionEnd(ContikiRadio.java:218)
        at org.contikios.cooja.radiomediums.AbstractRadioMedium.lambda$new$0(AbstractRadioMedium.java:190)
        at org.contikios.cooja.radiomediums.AbstractRadioMedium$$Lambda/0x00007f44b9f1cde0.accept(Unknown Source)
        at org.contikios.cooja.util.EventTriggers.trigger(EventTriggers.java:111)
        at org.contikios.cooja.contikimote.interfaces.ContikiRadio.doActionsAfterTick(ContikiRadio.java:340)
        at org.contikios.cooja.contikimote.ContikiMote.execute(ContikiMote.java:131)
        at org.contikios.cooja.motes.AbstractWakeupMote$1.execute(AbstractWakeupMote.java:67)
        at org.contikios.cooja.Simulation.lambda$new$0(Simulation.java:226)
        at org.contikios.cooja.Simulation$$Lambda/0x00007f44b9f20af8.run(Unknown Source)
        at java.base/java.lang.Thread.runWith(Thread.java:1596)
        at java.base/java.lang.Thread.run(Thread.java:1583)
```

alexrayne, 2024-01-07:
Are your simulation timeline keeps moment of start, when crush?
As i remember messages-list, and timeline have limit of events amount. Is
that limit has bounded?

ArshdeepSingh18, 2024-01-07:
Hello,

I am sorry I am new to Cooja, I am not sure I understand it. Can you please explain it a bit more.

Thanks

pjonsson, 2024-01-08:
@ArshdeepSingh18 is this a long running simulation? You can try to give the JVM a larger heap, change the `2048M` on the line `applicationDefaultJvmArgs = ['-Xms400M', '-Xmx2048M',`in `tools/cooja/build.gradle` to something larger.

ArshdeepSingh18, 2024-01-08:
@pjonsson Thanks for the update. 

Yes, My experiment requires from 42 nodes to 64 nodes, all `z1`. The packet exchange between nodes to the sink is every 60 seconds. Yes, For my experiment I require it to run for longer periods of time.

I did change the Xms to `8G` and Xmx to `8G`, but after half an hour into the simulation, it still throws an `OutOfMemory` error.

pjonsson, 2024-01-10:
@ArshdeepSingh18 based on what you have written, my guess is that Cooja is holding on to a lot of memory for a long time, and then the heap gets full and Cooja crashes.

This is obviously not desirable behavior, but I'm not sure how to fix it; if we drop some of the data the timeline will not be complete.

alexrayne, 2024-01-10:
As i remember, timeline have a limit of events, similar that for mote
messages list.
That limit surely works on old cooja - 2years ago. Maybe that has broken
since later rework.

ArshdeepSingh18, 2024-01-12:
@pjonsson Sorry for the late reply, I completely understand that, do you think there can be a way around it?