contiki-ng/contiki-ng #2839: Process suddenly not running after several iterations (CC13x2)

micw, 2024-01-09:
Hello,

I try to debug a strange issue. I have a firmware built on contiki-ng. It makes heavy use of the process/event concept. The firmware wakes up periodically, starts process which communicates with events, stops the processes and goes to sleep. Then it all repeats.

Each process thread has a similar structure:

```
PROCESS_THREAD(my_process_a, ev, data)
{
    PROCESS_BEGIN();
    while(1)
    {
        PROCESS_YIELD();
     // do some stuff
    }
    PROCESS_END();
}
```

Processes are started with `process_start(&my_process_a, NULL);` and stopped with `process_exit(&my_process_a);`

`my_process_a` sends some events to `my_process_b`. I log the state like this in `my_process_a`:

```
        LOG_INFO("my_process_a 1, my_process_b running %i\n",process_is_running(&my_process_b));
        int result=process_post(&my_process_b, custom_event.some_event_name NULL);
        LOG_INFO("my_process_a 2, my_process_b running %i, result %i\n",process_is_running(&brick_lz_sub_task),result);
```

In `my_process_b` I do a log before and after `PROCESS_YIELD();`

I observe the following behavior. For a lot of sleep/wake-up/sleep cycle, all works well. "my_process_b running" is 1 when the event is posted, I can see in the logs that `PROCESS_YIELD();` in `my_process_b` wakes up, does it's job, the loop starts and I see the log before the yield (so process B is waiting for an event).

But after _exactly_ 52 cycles, it misbehaves. I can see that process b is running, handles one event and I see the log before the yield. That's the last thing I see from this process. In process A, I can see that "my_process_b running" is suddenly 0, so the process stopped. I have searched all the code. There is only one single `process_exit(&my_process_b);` with a log statement and this exit is not called.

Is there anything else that could stop a process and remove it from the process list?

Thank you in advance,
Michael.

micw, 2024-01-09:
There are 2 more findings:

1.
`my_process_a` sends some events to `my_process_b`. `my_process_b` does then some stuff and sends back an event to `my_process_a`. In the error case, `my_process_b`is running in the moment it sends it's events to `my_process_a`. But when `my_process_a` receives the event (via process_yield), `my_process_b` is not running anymore. So  `my_process_b` is stoppen somewhere before  `my_process_a` processes the event.

2.
 `my_process_a`  and  `my_process_b` are never terminated before the plattform goes to sleep. This is because the termination function calls a process_post_synch that never returns, prir to going to sleep. May it be that some "non terminated" processes adds up until something bad™ happens?

micw, 2024-01-09:
To verify/exclude 2., I changed the synch event to asynch. Now the process is stopped correctly before going to sleep - but the behavior remains the same. Process get somehow stoppped in the 52th iteration...

micw, 2024-01-10:
I have enabled process debugging and saw that the process is actually exited. Since I have ensured that our code does not exit the process, I searched the contiki-ng codebase for exit_process. I found 2 occurences. One in process_exit (expected) and one in call_process (quite unexpected).

So I added a log in call_process at https://github.com/contiki-ng/contiki-ng/blob/develop/os/sys/process.c#L206 I think there should be a debug log by default. Seems my process B is terminated directly after it received an event from process A. I'm investigating further...

micw, 2024-01-10:
I found the issue. We have defined and allocated custom events as described in the wiki:

```
struct custom_events_t
{
    process_event_t custom_event1;
    process_event_t custom_event2;
    process_event_t custom_event3;
    process_event_t custom_event4;
};
```

```
PROCESS_THREAD(my_process_a, ev, data)
{
    PROCESS_BEGIN();

    custom_events_t.custom_event1 = process_alloc_event();
    custom_events_t.custom_event2 = process_alloc_event();
    custom_events_t.custom_event3 = process_alloc_event();
    custom_events_t.custom_event4 = process_alloc_event();
```

I have logged the int value of the custom event on each call. Seems that the number starts with some value (179 in the case of custom_event1) and increases by 4 on each wakeup when the process is restarted. So after 52 iterations it ran-over and reached 131 - the event code for "PROCESS_EVENT_EXIT". 

I'm not sure if that's a mistake on our side (because the event is allocated that often). But I'm pretty sure that process_alloc_event() should never return one of the "reserved" event codes. I'd even say it should log a warning if the number of available events exceeds.

micw, 2024-01-10:
I have changed the title. I think the following needs to be checked:

1. Docs, https://docs.contiki-ng.org/en/develop/doc/programming/Processes-and-events.html#user-defined-events
Make clear when to allocate the event ID. Should it be when the process is started? The examples suggest this (see https://github.com/contiki-ng/contiki-ng/blob/develop/examples/platform-specific/cc26x0-cc13x0/cc26x0-web-demo/httpd-simple.c#L1331). But a process might be started multiple times after it was stopped, that's allowed by the spec.

2. Should process_alloc_event() overflow silently?

3. Should process_alloc_event() return reserved event numbers?

2+3 causes very hard-to-debug issues. Took me more than a day to figure it out :-(

alexrayne, 2024-01-10:
Alloc_event not intended for allocating events constantly or infinite. It
replacer of manual event allocation. You have miss-use it.
As for your design - you missuse event: use it as a counter. If you neen
counter in event, use attached data*

micw, 2024-01-10:
@alexrayne Clear to me _after_ I investigated the issue. But from the docs, the examples and the code comments, I'd say I used it as the framework suggests.

micw, 2024-01-10:
The same Issue might also be triggered by contiki-ng's internal stuff.
Examples:
* sensors.c
* tcpip.c

Each allocates the event ID when the process is started. So also when I restart sensors 255 times, event might be 131 that stops the process.

alexrayne, 2024-01-11:
Eah, looks there (tcpip, sensors) is a bugs. Events should allocates once -
if not allocated. Right after reset it`s value = 0, so there should be a
chek vs 0-value

alexrayne, 2024-01-11:
Also,  tcpip and sensors processes run once at a start, as i remember, wake
from sleep not restarts them. So events overun not happens for that modules.

arurke, 2024-07-06:
@micw Did https://github.com/contiki-ng/contiki-ng/pull/2866 address your concerns?

micw, 2024-07-06:
Absolutely, yes. Thank you a lot!

closed 2024-07-06 by micw