contiki-ng/contiki-ng #2848: Problem when starting Cooja simulator

dfonovic, 2024-01-24:
I got this error when I try to start Cooja simulator from Docker conatainer:
 
root@9e6c9da70970:~/contiki-ng# cooja
Starting a Gradle Daemon, 1 incompatible and 1 stopped Daemons could not be reused, use --status for details

FAILURE: Build failed with an exception.

* What went wrong:
Could not open settings generic class cache for settings file '/home/user/contiki-ng/tools/cooja/settings.gradle' (/root/.gradle/caches/8.1.1/scripts/8gn3g19clydc2xzaxxc8hxbnn).
> BUG! exception in phase 'semantic analysis' in source unit '_BuildScript_' Unsupported class file major version 65

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 8s
root@9e6c9da70970:~/contiki-ng#

pjonsson, 2024-01-26:
Use the `user` user inside the container, not root.

yvshudrenko, 2024-06-21:
Might be related to Java 21 being the default, while Gradle 8.1.1 only supports Java 20.
```bash
user@f3d24c51036a:~/contiki-ng/tools/cooja$ java --version
java 21.0.3 2024-04-16 LTS
Java(TM) SE Runtime Environment Oracle GraalVM 21.0.3+7.1 (build 21.0.3+7-LTS-jvmci-23.1-b37)
Java HotSpot(TM) 64-Bit Server VM Oracle GraalVM 21.0.3+7.1 (build 21.0.3+7-LTS-jvmci-23.1-b37, mixed mode, sharing)
user@f3d24c51036a:~/contiki-ng/tools/cooja$ 
```

pjonsson, 2024-06-21:
The Java 21 issue with Gradle 8.1.1 sounds plausible, try with an older contiki-ng image and see if that works better.

yvshudrenko, 2024-07-24:
Recent images had the same issue, tried one randomly from a bit longer ago - 8f38c24bc0c292a0115b36eb9aad1efa4dddfd5aeecc12693c30cc170fc65a54, it works. There are still some hiccups related to X11 with M1 Mac, but it's a different story.

Not sure what's the deal here in general with macOS, seems to work fine on WSL with latest contiki image.

tenllado, 2024-10-02:
I am trying to get cooja running on linux using docker. I followed the getting started guide to install the docker image.
I tried with contiker/contiki-ng:latest , and some older images, including 8f38c24bc0c292a0115b36eb9aad1efa4dddfd5aeecc12693c30cc170fc65a54. None of them works for me.

On the contiker/contiki-ng:latest I get:

```
user@2b26d50817cf:~/contiki-ng/tools/cooja$ ./gradlew run
Downloading https://services.gradle.org/distributions/gradle-8.10-bin.zip

Exception in thread "main" java.net.UnknownHostException: services.gradle.org
	at java.base/sun.nio.ch.NioSocketImpl.connect(NioSocketImpl.java:567)
	at java.base/java.net.SocksSocketImpl.connect(SocksSocketImpl.java:327)
	at java.base/java.net.Socket.connect(Socket.java:751)
	at java.base/sun.security.ssl.SSLSocketImpl.connect(SSLSocketImpl.java:304)
	at java.base/sun.net.NetworkClient.doConnect(NetworkClient.java:178)
	at java.base/sun.net.www.http.HttpClient.openServer(HttpClient.java:531)
	at java.base/sun.net.www.http.HttpClient.openServer(HttpClient.java:636)
	at java.base/sun.net.www.protocol.https.HttpsClient.<init>(HttpsClient.java:264)
	at java.base/sun.net.www.protocol.https.HttpsClient.New(HttpsClient.java:377)
	at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.getNewHttpClient(AbstractDelegateHttpsURLConnection.java:193)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1237)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1123)
	at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:179)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1675)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1599)
	at java.base/sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:223)
	at org.gradle.wrapper.Install.forceFetch(SourceFile:2)
	at org.gradle.wrapper.Install$1.call(SourceFile:8)
	at org.gradle.wrapper.GradleWrapperMain.main(SourceFile:67)
```

Java version in this container is:

```
user@2b26d50817cf:~/contiki-ng/tools/cooja$ java -version
java version "21.0.4" 2024-07-16 LTS
Java(TM) SE Runtime Environment Oracle GraalVM 21.0.4+8.1 (build 21.0.4+8-LTS-jvmci-23.1-b41)
Java HotSpot(TM) 64-Bit Server VM Oracle GraalVM 21.0.4+8.1 (build 21.0.4+8-LTS-jvmci-23.1-b41, mixed mode, sharing)

```
Same output for the contiker/contiki-ng:8f38c24bc0c292a0115b36eb9aad1efa4dddfd5aeecc12693c30cc170fc65a54:

```
user@ec36d7b5ed99:~/contiki-ng/tools/cooja$ ./gradlew run
Downloading https://services.gradle.org/distributions/gradle-8.10-bin.zip

Exception in thread "main" java.net.UnknownHostException: services.gradle.org
	at java.base/sun.nio.ch.NioSocketImpl.connect(NioSocketImpl.java:567)
	at java.base/java.net.SocksSocketImpl.connect(SocksSocketImpl.java:327)
	at java.base/java.net.Socket.connect(Socket.java:633)
	at java.base/sun.security.ssl.SSLSocketImpl.connect(SSLSocketImpl.java:304)
	at java.base/sun.net.NetworkClient.doConnect(NetworkClient.java:178)
	at java.base/sun.net.www.http.HttpClient.openServer(HttpClient.java:531)
	at java.base/sun.net.www.http.HttpClient.openServer(HttpClient.java:636)
	at java.base/sun.net.www.protocol.https.HttpsClient.<init>(HttpsClient.java:266)
	at java.base/sun.net.www.protocol.https.HttpsClient.New(HttpsClient.java:380)
	at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.getNewHttpClient(AbstractDelegateHttpsURLConnection.java:193)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1242)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1128)
	at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:179)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1665)
	at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1589)
	at java.base/sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:224)
	at org.gradle.wrapper.Install.forceFetch(SourceFile:2)
	at org.gradle.wrapper.Install$1.call(SourceFile:8)
	at org.gradle.wrapper.GradleWrapperMain.main(SourceFile:67)
```

Java version in this container is:

```
user@ec36d7b5ed99:~/contiki-ng/tools/cooja$ java -version
openjdk version "17.0.7" 2023-04-18
OpenJDK Runtime Environment GraalVM CE 17.0.7+7.1 (build 17.0.7+7-jvmci-23.0-b12)
OpenJDK 64-Bit Server VM GraalVM CE 17.0.7+7.1 (build 17.0.7+7-jvmci-23.0-b12, mixed mode, sharing)
```

Is the problem really related to the java version?

Am I doing something wrong to start cooja from the container? Do I have to do/configure something else?

Any help appreciated.

pjonsson, 2024-10-02:
@tenllado for some reason, your container cannot find services.gradle.org. I can download https://services.gradle.org/distributions/gradle-8.10-bin.zip just fine on my machine, so your problem has nothing to do with the Java version. It's also unlikely to be a Cooja bug, I suggest looking into DNS- or firewall-settings on the host machine.

tenllado, 2024-10-02:
@pjonsson thank you so much, you are right, it is a problem from the container. I solved it using 8.8.8.8 as dns for dockerd but now it rises an error when opening the window:

```
...
Authorization required, but no authorization protocol specified
Exception in thread "main" java.awt.AWTError: Can't connect to X11 window server using ':0' as the value of the DISPLAY variable
...
```

I am running the container with the recommended command in the getting started guide:

```
docker run --privileged --sysctl net.ipv6.conf.all.disable_ipv6=0 --mount type=bind,source=$CNG_PATH,destination=/home/user/contiki-ng -e DISPLAY=$DISPLAY -e LOCAL_UID=$(id -u $USER) -e LOCAL_GID=$(id -g $USER) -v /tmp/.X11-unix:/tmp/.X11-unix -v /dev/bus/usb:/dev/bus/usb -ti --rm contiker/contiki-ng

```

I am sure it is a problem of alowing docker to open windows, just I do not know what to do. I am using gnome3 with wayland, just in case that matters. Any help would be appreciated. Thanks.

tenllado, 2024-10-07:
For anyone interested, the solution was to run before the command xhost +SI:localuser:$(id -un).