contiki-ng/contiki-ng #2853: Error when using cooja with no-gui, test failed after timeout.

YanboChenA, 2024-02-05:
It works when I use cooja with gui, and run the simulation without script. When I try use no-gui to run the csc file, or activate the script in cooja with gui, the same error occur. It's running before timeout.

```
> Task :run
INFO  [main] [Cooja.java:1145] - Loading ../../examples/6tisch/simple-node-udp/cooja.csc random seed: null
> make -j4 node.cooja TARGET=cooja
  LD        build/cooja/node.cooja
  CP        build/cooja/node.cooja --> node.cooja
INFO  [main] [LogScriptEngine.java:231] - Script timeout in 100000 ms
INFO  [main] [LogScriptEngine.java:231] - Script timeout in 100000 ms
INFO  [sim] [LogScriptEngine.java:320] -  5% completed, 11.9 sec remaining
INFO  [sim] [LogScriptEngine.java:320] -  5% completed, 10.4 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 10% completed,  8.9 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 10% completed,  8.2 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 15% completed,  7.5 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 15% completed,  7.0 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 20% completed,  6.7 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 20% completed,  6.3 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 25% completed,  6.0 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 25% completed,  5.7 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 30% completed,  5.4 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 30% completed,  5.2 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 35% completed,  4.9 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 35% completed,  4.8 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 40% completed,  4.5 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 40% completed,  4.3 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 45% completed,  4.1 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 45% completed,  4.0 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 50% completed,  3.7 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 50% completed,  3.6 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 55% completed,  3.4 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 55% completed,  3.3 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 60% completed,  3.0 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 60% completed,  3.0 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 65% completed,  2.7 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 65% completed,  2.6 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 70% completed,  2.3 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 70% completed,  2.2 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 75% completed,  1.9 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 75% completed,  1.9 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 80% completed,  1.5 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 80% completed,  1.5 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 85% completed,  1.2 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 85% completed,  1.1 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 90% completed,  0.8 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 90% completed,  0.8 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 95% completed,  0.4 sec remaining
INFO  [sim] [LogScriptEngine.java:320] - 95% completed,  0.4 sec remaining
INFO  [sim] [LogScriptEngine.java:294] - Timeout event @ 100000000
INFO  [sim] [Simulation.java:450] - Runtime: 7648 ms. Simulated time: 100000 ms. Speedup: 13.07531380753138
ERROR [main] [Cooja.java:1173] - TEST ../../examples/6tisch/simple-node-udp/cooja.csc FAILED

ERROR [main] [Cooja.java:1181] - Failed tests:
../../examples/6tisch/simple-node-udp/cooja.csc seed: null

> Task :run FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':run'.
> Process 'command '/home/yanbo/.gradle/jdks/eclipse_adoptium-21-amd64-linux/jdk-21.0.1+12/bin/java'' finished with non-zero exit value 1

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 11s
3 actionable tasks: 1 executed, 2 up-to-date
```

pjonsson, 2024-02-09:
I don't have an `examples/6tisch/simple-node-udp` in my tree, so you're running an example constructed by someone else. Does it work if you make the timeout longer?

PatrykFonseca, 2024-10-17:
I had the same issue, the problem is that in .csc file the TIMEOUT is not set. Every .csc file running on cooja without gui needs to have a timout setting.

Something like: ` <script>TIMEOUT(360000);</script>`

Try running this file:
./gradlew run --args='--no-gui ../../tests/07-simulation-base/19-cooja-rpl-tsch.csc'