contiki-ng/contiki-ng #2910: Border Router, or my brain have a bug

rahav, 2024-05-17:
After a couple of debugging days I am at a loss. I reported on #2906 and #2907 and this morning tried the native border router, which also does not work. I am certain that I am doing something wrong because I cannot imagine an IoT OS without something such as the border router. If it really does not work, is there an alternative to communicate with the outside world?

Can someone please take a look, or let me know what to try next?

My experience with the [embedded BR](https://github.com/contiki-ng/contiki-ng/discussions/2906). Following is the experience with the native BR:
1. Program nRF52840 DK with `examples/slip-radio`
2. Program another nRF52840 DK with `examples/hello-world`.
3. `make TARGET=native` in `examples/rpl-border-router/`
4. Exit the Contiki Docker container, and on the host OS, run:

```
sudo examples/rpl-border-router/build/native/border-router.native fd00::1/64 -s /dev/ttyACM0
[INFO: Main      ] Starting Contiki-NG-288b134d6-dirty
[INFO: Main      ] - Routing: RPL Classic
[INFO: Main      ] - Net: sicslowpan
[INFO: Main      ] - MAC: br-mac
[INFO: Main      ] - 802.15.4 PANID: 0xabcd
[INFO: Main      ] - 802.15.4 Default channel: 26
[INFO: Main      ] Node ID: 1800
[INFO: Main      ] Link-layer address: 0102.0304.0506.0708
[INFO: Main      ] Tentative link-local IPv6 address: fe80::302:304:506:708
[INFO: Native    ] Added global IPv6 address fd00::302:304:506:708
[INFO: RPL BR    ] Contiki-NG Border Router started
[INFO: BR        ] RPL-Border router started
********SLIP started on /dev/ttyACM0'' opened tun device /dev/tun0''
ifconfig tun0 inet hostname up
ifconfig tun0 add fd00::1/64
ifconfig tun0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 127.0.1.1  netmask 255.255.255.255  destination 127.0.1.1
        inet6 fd00::1  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::6417:e996:4024:8ede  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

[INFO: BR        ] Setting prefix fd00::1
[INFO: BR        ] Server IPv6 addresses:
[INFO: BR        ]   fd00::f6ce:3676:a2a7:695
[INFO: BR        ]   fe80::f6ce:3676:a2a7:695
```

Above looks as expected, however, pinging the BR in another terminal window on the host OS yields nothing:
```
ping6 fd00::f6ce:3676:a2a7:695
PING fd00::f6ce:3676:a2a7:695(fd00::f6ce:3676:a2a7:695) 56 data bytes
^C
--- fd00::f6ce:3676:a2a7:695 ping statistics ---
5 packets transmitted, 0 received, 100% packet loss, time 4078ms
```

I know that the nRF52840 requires hardware flow control, so I added `-H` argument:
```
sudo examples/rpl-border-router/build/native/border-router.native fd00::1/64 -H -s /dev/ttyACM0
```
but it did not work either.

rahav, 2024-05-17:
I switched from the nRF52840 DK to the Dongle. This allows the Embedded BR work.

The native BR continues to not work after the switch.

The BR is an important component in the Contiki stack. May I ask to prioritize a fix so it works on the DK as embedded, as well as native BR? I will be happy to contribute with some guidance.

rahav, 2024-05-28:
Anyone please. Does the Border Router works with the 52840 DK for anyone, or is this a known bug? Is there a workaround?

rahav, 2024-06-24:
It problem seems to be more intricate. The DK will work as a BR if we enable the native USB port of the nRF52840 SoC, using the `NRF52840_NATIVE_USB=1` argument, when compiling the code.   

Was that the intent @Yagoor ? (I thought that enabling this was optional).

maxime-ole-bucher, 2024-07-08:
I also have the same problem (nrf52840dk).

When I increase the logging level, I observe that the pings from the host are received, but it seems they are never processed.

I ran :
```
sudo ../../tools/serial-io/tunslip6 -v5 -L -s /dev/ttyACM0 fd00::1/64
```

I got this output for the border router :
```
11:25:36 ********SLIP started on ``/dev/ttyACM0''
0000.011 opened tun device ``/dev/tun0''
0000.011 ifconfig tun0 inet `hostname` mtu 1500 up
0000.017 ifconfig tun0 add fd00::1/64
0000.020 ifconfig tun0 add fe80::0:0:0:1/64
0000.023 ifconfig tun0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 127.0.1.1  netmask 255.255.255.255  destination 127.0.1.1
        inet6 fd00::1  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::25e0:c3b5:acb8:e442  prefixlen 64  scopeid 0x20<link>
        inet6 fe80::1  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

0000.027 Packet from TUN of length 48 - write SLIP
0000 60 00 00 00 00 08 3a ff fe 80 00 00 00 00 00 00 25 e0 c3 b5 ac b8 e4 42 ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 02 85 00 02 a6 00 00 00 00
0000.641 [INFO: BR        ] Waiting for prefix
0000.642 *** Address:fd00::1 => fd00:0000:0000:0000
0001.641 [INFO: BR        ] Waiting for prefix
0001.645 [INFO: BR        ] Server IPv6 addresses:
0001.649 [INFO: BR        ]   fd00::f6ce:36d4:fb80:ef32
0001.653 [INFO: BR        ]   fe80::f6ce:36d4:fb80:ef32
```

After pinging the border router from the host I got this (on the board) : 
```
0004.484 Packet from TUN of length 48 - write SLIP
0000 60 00 00 00 00 08 3a ff fe 80 00 00 00 00 00 00 ba df 55 41 fd 5b 1a ba ff 02 00 00 00 00 00 00 00 00 00 00 00 00 00 02 85 00 55 00 00 00 00 00
0005.997 Packet from TUN of length 104 - write SLIP
0000 60 0f e5 39 00 40 3a 40 fd 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 fd 00 00 00 00 00 00 00 f6 ce 36 d4 fb 80 ef 32 80 00 6b 23 66 51 00 01 07 ad 8b 66 00 00 00 00 3d cf 0c 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
0007.044 Packet from TUN of length 104 - write SLIP
0000 60 0f e5 39 00 40 3a 40 fd 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 fd 00 00 00 00 00 00 00 f6 ce 36 d4 fb 80 ef 32 80 00 ed 6b 66 51 00 02 08 ad 8b 66 00 00 00 00 b9 85 0d 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
```

I can't see exactly where it is blocking.