contiki-ng/contiki-ng #2921: bin/hex files missing for arm on contiki-ng 4.9

micw, 2024-06-14:
Hello,

with contiki-ng 4.8 I got bin and hex files from "arm" builds. The bin was the file I could easily flash. On contiki-ng 4.9 these files are not build anymore.

I bisected and found the following commit as cause (which indeed states this change in the comment):

```
130c672035647df194d305fc3385065939146421 is the first bad commit
commit 130c672035647df194d305fc3385065939146421 (HEAD)
Author: Peter A. Jonsson <jonsson.peter.a@gmail.com>
Date:   Thu Jul 21 21:40:41 2022 +0200

    arm: remove false dependencies
    
    The hex and bin files are not required
    for the target according to the comment,
    so stop building them.
    
    This is preparation to remove the customrules
    for Cortex-M, but it also saves IO load
    on the IO-limited CI machines.

 arch/cpu/arm/Makefile.arm | 4 +---
 1 file changed, 1 insertion(+), 3 deletions(-)
```

Why was this change made? Why are these files considered to be a "false dependency"? What is "the comment" that says these files are not required?

I propose to revert the commit and (optionally) make the build configurable.

micw, 2024-06-14:
Commit is here: https://github.com/contiki-ng/contiki-ng/commit/130c672035647df194d305fc3385065939146421

There is the comment: https://github.com/contiki-ng/contiki-ng/commit/130c672035647df194d305fc3385065939146421#diff-0cab189fc04667ad50e4aef1e74c5057d7502a0b85c858a536df187bb8be12ceL96-R96

It is unclear who "we" is and why it's not needed but built anyway.

M-Creeper, 2024-06-24:
I also really need these files, simply adding back the lines does not seem to do the trick for me.. how can i get an .elf file? I really need it and am starting to get desperate...

arurke, 2024-06-29:
> how can i get an .elf file?

The <application>.<target> file (hello-world.simplelink, hello-world.cc26x0-cc13x0, etc.) is an .elf file.

Seems #2084 was a bit hastily merged. The [documentation](https://docs.contiki-ng.org/en/develop/doc/platforms/cc26x0-cc13x0.html?highlight=elf#how-to-program-your-device) also states there should be a .bin and .hex file. What do you think @pjonsson , any arguments against reverting (I see you mention this is a stepping-stone to removing customrules)?

pjonsson, 2024-06-30:
> It is unclear who "we" is and why it's not needed but built anyway.

The comment is probably written by someone whose first language isn't English. The comment means "%.$(TARGET) can be produced without %.bin and %.hex, but build them anyways.". I don't know the motivation for building the files despite them not being required.

> The [documentation](https://docs.contiki-ng.org/en/develop/doc/platforms/cc26x0-cc13x0.html?highlight=elf#how-to-program-your-device) also states there should be a .bin and .hex file. What do you [...]

It's not isolated to the specific commit, but as a general description, the build system is effectively trying to guess what the user wants from incomplete input, and that complicates the build system and makes the build slower. This might have been best practice at the time, but it's difficult to maintain and not how modern build systems work. The complicated build system is only an issue for Contiki-NG developers, but slow builds are an issue for everyone. Why can't the one who wants a hex or bin file build them explicitly with `make TARGET=<target> build/board/myfile.{bin,hex}`?

The documentation is obviously wrong after the commit, so that should have been updated.

arurke, 2024-07-01:
> Why can't the one who wants a hex or bin file build them explicitly with make TARGET=<target> build/board/myfile.{bin,hex}?

Would it be sensible to toggle it via a flag? Something agnostic to the platform and formats such as `MAKE_BUILD_ALL_FORMATS` - and it would be up to the platforms to output whatever they can when this is set. That way the formats are more accessible, and it would be more user-friendly to those that relied on the old behavior.