contiki-ng/contiki-ng #2929: For the dockerfile should you use ADD over wget?

mofosyne, 2024-06-30:
@arurke while I was trying to figure out what's wrong with the dockerfile in https://github.com/contiki-ng/contiki-ng/pull/2680 I came across an article about how ADD may be better than wget in terms of dealing with caching (because it took ages to download/build the image)

So instead of 

```dockerfile
# Install nRF Util+nRF Command Line tools.
RUN wget -nv -O /usr/local/bin/nrfutil https://developer.nordicsemi.com/.pc-tools/nrfutil/x64-linux/nrfutil && \
  chmod 755 /usr/local/bin/nrfutil && \
  wget -nv https://nsscprodmedia.blob.core.windows.net/prod/software-and-other-downloads/desktop-software/nrf-command-line-tools/sw/versions-10-x-x/10-22-0/nrf-command-line-tools_10.22.0_amd64.deb && \
  apt-get -qq -y --no-install-recommends install ./nrf-command-line-tools_10.22.0_amd64.deb > /dev/null && \
  apt-get -qq -y --no-install-recommends install /opt/nrf-command-line-tools/share/JLink_Linux_V780c_x86_64.deb > /dev/null && \
  rm -f *.deb *.tar.gz /opt/nrf-command-line-tools/share/JLink_*deb && \
  apt-get -qq clean
```

Could we rewrite it as below (untested)?

```dockerfile
ADD https://developer.nordicsemi.com/.pc-tools/nrfutil/x64-linux/nrfutil ./ # ./nrfutil
RUN mv ./nrfutil /usr/local/bin/nrfutil
RUN chmod 755 /usr/local/bin/nrfutil && \
  wget -nv https://nsscprodmedia.blob.core.windows.net/prod/software-and-other-downloads/desktop-software/nrf-command-line-tools/sw/versions-10-x-x/10-22-0/nrf-command-line-tools_10.22.0_amd64.deb && \
  apt-get -qq -y --no-install-recommends install ./nrf-command-line-tools_10.22.0_amd64.deb > /dev/null && \
  apt-get -qq -y --no-install-recommends install /opt/nrf-command-line-tools/share/JLink_Linux_V780c_x86_64.deb > /dev/null && \
  rm -f *.deb *.tar.gz /opt/nrf-command-line-tools/share/JLink_*deb && \
  apt-get -qq clean
```

FYI I've made https://briankhuu.com/tools/docker-convert-to-add/ to hopefully make the process easier, especially considering the annoying factor of ADD not allowing you to rename the output file...

pjonsson, 2024-06-30:
I'm not sure what you mean by not allowing you to rename the output file, https://docs.docker.com/reference/dockerfile/#add shows a destination argument, and your code example in this issue even uses the destination argument.

The Contiki-NG image is meant to be pulled by users, and it's rebuilt a couple of times per month, so shorter image build times are nice but not essential. The image size is essential though. The image is meant to be used interactively so size being essential still doesn't mean we should remove bash in the interest of saving space, but we shouldn't waste space needlessly. Each RUN statement creates a new layer in the resulting image, and each layer takes up space.

The example will give one binary diff from the ADD, a second binary diff from the RUN that moves the file, and a third binary diff from the chmod. And yes, those layers could be squashed by some third-party tool after the image is built, but that complicates the build process.

For the particular case of nrfutil, I guess that is a binary file, so that could probably be ADDed straight to /usr/local/bin/nrfutil with the right name/owner/mode, and that ADD could be early in the Dockerfile so it would remain cached.

The the deb/zip/etc files are not in the final image, so it's better to keep those within a RUN statement for size reasons.

As for (Docker) caching: there are bugs in Docker so it sometimes loses track of blobs, and that causes build failures with a long hex string not being found. If you want to use cache locally and are prepared to deal with the issues that is fine, but for "upstream" I don't think the time savings from shorter image build times can outweigh the cost of the confusion/support time for those problems right now.

arurke, 2024-06-30:
@mofosyne Also, if I got it right, the [accepted answer](https://stackoverflow.com/a/75757239) in the article/thread you are referring to corroborates what @pjonsson mentions. Looks to me as poor trade-off.

mofosyne, 2024-06-30:
Ah well. I wish there was a better way of caching but I see what you mean now. At least versioning in filenames in url would help here. But of course if it's /latest in then that's not going to always work

closed 2024-07-07 by mofosyne