contiki-ng/contiki-ng #2953: Clarification on Initial ETX Calculation from RSSI in ETX Metric

ceylanim, 2024-09-09:
Hello,

I’ve come across the logic for calculating the initial ETX value from an RSSI value, and I have a question about the way it’s handled for low RSSI values.

According to the comments:

When RSSI ≥ RSSI_HIGH, the default ETX is used.
When RSSI_LOW < RSSI < RSSI_HIGH, ETX is a linear function of RSSI.
When RSSI ≤ RSSI_LOW, the minimal initial ETX is used.
My understanding is that ETX is a cost metric, and a higher cost should correspond to a higher ETX value. However, in the case where RSSI is below RSSI_LOW (e.g., -90), the minimal ETX is used, which seems counterintuitive to me. Shouldn’t the ETX value be higher when the RSSI is very low, considering that the cost of transmission would likely increase?
Additionally, I conducted some experiments using log_info on the application, and I noticed that when RSSI is -50 and -60, the ETX for the -50 RSSI is actually higher than for -60 RSSI. This also seems counterintuitive, as I would expect the ETX to be lower for better (higher) RSSI values.
Could you please provide some clarification on this?
Thank you!

ceylanim, 2024-09-09:
[Test on an online compiler](https://www.programiz.com/online-compiler/0Tjl9gydrWec1)
The test calculates the initial ETX from RSSI values. For an RSSI of -90, the ETX is 256, and for an RSSI of -60, the ETX is 384. However, I was expecting the ETX for -60 to be smaller than for -90, as -60 indicates better signal strength.

Is there an issue with my understanding, or could there be a problem with the ETX calculation logic?

Thank you!

ayushj1912, 2025-07-23:
Even I have encountered same issue