contiki-ng/contiki-ng #2991: etimer bugs？

yanxxx08cn, 2024-12-10:
I encountered some problems when using etimer to drive tasks. I am not sure if it is a bug. The phenomenon is that after contiki-ng runs for a period of time, the etimer pointers of some random tasks will disappear from the timerlist, causing the corresponding tasks to not run. I did not call process_exit to exit the process. Is there anything else that may cause the process to exit?

alexrayne, 2024-12-10:
as i remember, last year was acepted optimisation of etimer, you can try take early source of etimer from history and check with it.
former etimer was a bit simpler.

yanxxx08cn, 2024-12-10:
> as i remember, last year was acepted optimisation of etimer, you can try take early source of etimer from history and check with it. former etimer was a bit simpler.

> as i remember, last year was acepted optimisation of etimer, you can try take early source of etimer from history and check with it. former etimer was a bit simpler.
I checked the code of etimer_process. When it happened before, I forgot to check whether the process in process_list still existed. At present, it is judged that after timer_expired in etimer_process, the process_post function is used to send the event, but the target task is not executed, resulting in the etimer_set in the target task unable to add its own etimer to timerlist, thus causing the task to fail to run. Now I am still simulating and waiting for the situation to reappear.

yanxxx08cn, 2024-12-10:
I checked the code of etimer_process. When it happened before, I forgot to check whether the process in process_list still existed. At present, it is judged that after timer_expired in etimer_process, the process_post function is used to send the event, but the target task is not executed, resulting in the etimer_set in the target task unable to add its own etimer to timerlist, thus causing the task to fail to run. Now I am still simulating and waiting for the situation to reappear.

yanxxx08cn, 2024-12-10:
> as i remember, last year was acepted optimisation of etimer, you can try take early source of etimer from history and check with it. former etimer was a bit simpler.

I checked the code of etimer_process. When it happened before, I forgot to check whether the process in process_list still existed. At present, it is judged that after timer_expired in etimer_process, the process_post function is used to send the event, but the target task is not executed, resulting in the etimer_set in the target task unable to add its own etimer to timerlist, thus causing the task to fail to run. Now I am still simulating and waiting for the situation to reappear.

fozc, 2024-12-12:
Hello, I am currently experiencing a very similar issue. Although there are many tasks in the system, one of them stops running after a while. This task is polled using an etimer event. I haven't had any problems with tasks triggered by other events so far, but I am encountering this issue with tasks driven by etimer in two different projects.

alexrayne, 2024-12-12:
>> Is there anything else that may cause the process to exit?

process can finish natively by reach PROCESS_END() point, process_exit - is a tool for abnormal termination

kkrentz, 2024-12-12:
Here is a refactored version of etimer: https://github.com/kkrentz/contiki-ng/tree/etimer-pr. Maybe it accidentally solves your problem.

fozc, 2025-05-21:
> > Hello, I am currently experiencing a very similar issue. Although there are many tasks in the system, one of them stops running after a while. This task is polled using an etimer event. I haven't had any problems with tasks triggered by other events so far, but I am encountering this issue with tasks driven by etimer in two different projects.
> 
> I have also encountered the same problem. Have you solved it? Is there any good solution?

Hi, I believe it was a memory leak issue. I disabled all the tasks one by one and eventually caught the bug.

kkrentz, 2025-05-21:
What could be a problem is that the result of `process_poll` is not checked for errors.

ChunhuiYan-0121, 2025-05-23:
> Hi, I believe it was a memory leak issue. I disabled all the tasks one by one and eventually caught the bug.

Hello, memory leak? I am using msp430, and I don't see any memory allocation in the code. I have reproduced it before. For example, if I set a 500ms etimer timer, if I set it to 100ms before the 500ms timer is up, the timer will fail. This is one situation that will cause the etimer to fail, but there is also a probability that the timer will fail.

fozc, 2025-05-23:
> Hello, memory leak? I am using msp430, and I don't see any memory allocation in the code. I have reproduced it before. For example, if I set a 500ms etimer timer, if I set it to 100ms before the [...]

Hi,
I didn't write very clearly. I mean, my code had memory leak not the contiki-core.

ChunhuiYan-0121, 2025-05-23:
> > > > > Hello, I am currently experiencing a very similar issue. Although there are many tasks in the system, one of them stops running after a while. This task is polled using an etimer event. I haven't had any problems with tasks triggered by other events so far, but I am encountering this issue with tasks driven by etimer in two different projects.
> > > > 
> > > > 
> > > > I have also encountered the same problem. Have you solved it? Is there any good solution?
> > > 
> > > 
> > > Hi, I believe it was a memory leak issue. I disabled all the tasks one by one and eventually caught the bug.
> > 
> > 
> > Hello, memory leak? I am using msp430, and I don't see any memory allocation in the code. I have reproduced it before. For example, if I set a 500ms etimer timer, if I set it to 100ms before the 500ms timer is up, the timer will fail. This is one situation that will cause the etimer to fail, but there is also a probability that the timer will fail.
> 
> Hi, I didn't write very clearly. I mean, my code had memory leak not the contiki-core.
> 
> By the way, there should be a built-in delay function for processes in the contiki.

Thanks! I'll look at my code again.