contiki-ng/contiki-ng #3008: Make login fails on macOS

yvshudrenko, 2025-02-07:
Didn't flash any boards for quite some time (~year), then suddenly uploading works, [output.log](https://github.com/user-attachments/files/18710130/output.log), but not the login:

```
yevhenii@dhcp16 hello-world % gmake login PORT=/dev/tty.usbserial-1101             
using saved target 'openmote'
Running serialdump without rlwrap support.
Consider installing rlwarp in order to be able to use command history
gmake -C ../../tools/serial-io serialdump
gmake[1]: Entering directory '/Users/yevhenii/contiki-develop/tools/serial-io'
cc -I/opt/homebrew/include  -DCONTIKI=1 -DCONTIKI_TARGET_OPENMOTE=1 -DCONTIKI_TARGET_STRING=\"openmote\" -DCMSIS_DEV_HDR=\"cc2538_cm3.h\" -mcpu=cortex-m3 -mthumb -mabi=aapcs -mlittle-endian -Wall -std=c11 -ffunction-sections -fdata-sections -fno-strict-aliasing -fshort-enums -fomit-frame-pointer -Werror -Os -DCONTIKI_BOARD_OPENMOTE_B=1 -DCONTIKI_BOARD_STRING=\"openmote-b\" -DMAC_CONF_WITH_CSMA=1 -DNETSTACK_CONF_WITH_IPV6=1 -DROUTING_CONF_RPL_LITE=1  -I. -I../../arch/platform/openmote/. -I../../arch/platform/openmote/dev -I../../arch/platform/openmote/openmote-b -I../../arch/cpu/cc2538/. -I../../arch/cpu/cc2538/dev -I../../arch/cpu/cc2538/usb -I../../arch/cpu/cc2538/usb/common -I../../arch/cpu/cc2538/usb/common/cdc-acm -I../../arch/cpu/cc2538/../arm/cortex-m/cm3 -I../../arch/cpu/cc2538/../arm/CMSIS/CMSIS/Core/Include -I../../arch/cpu/cc2538/../arm/cortex-m -I../../arch/cpu/cc2538/../arm/. -I../../os -I../../arch -I../../os/sys -I../../os/lib -I../../os/dev -I../../os/net -I../../os/net/routing -I../../os/net/mac -I../../os/net/mac/framer -I../../os/storage -I../../os/services -I../../os/lib/newlib -I../../os/lib/dbg-io -I../../os/storage/cfs -I../../os/net/mac/csma -I../../os/net/ipv6 -I../../os/net/routing/rpl-lite  -I../.. -Wall -Werror -O2 serialdump.c tools-utils.c -o serialdump
clang: error: unsupported argument 'cortex-m3' to option '-mcpu='
clang: error: unsupported argument 'cortex-m3' to option '-mcpu='
gmake[1]: *** [Makefile:10: serialdump] Error 1
gmake[1]: Leaving directory '/Users/yevhenii/contiki-develop/tools/serial-io'
gmake: *** [../../Makefile.tools:13: ../../tools/serial-io/serialdump] Error 2
```

macOS 14.6.1 (23G93), all the simple things tried like checked master/develop, reinstalled all the dependencies, etc., don't have time/importance for deeper investigation

joakimeriksson, 2025-03-19:
I am trying the same thing on my Mac (with gmake 4.4.1) and it looks like this:
```
gmake login TARGET=openmote        
Running serialdump without rlwrap support.
Consider installing rlwarp in order to be able to use command history
gmake -C ../../tools/serial-io serialdump
gmake[1]: Entering directory '/Users/joakimeriksson/work/contiki-ng/tools/serial-io'
cc -Wall -Werror -O2 serialdump.c tools-utils.c -o serialdump
gmake[1]: Leaving directory '/Users/joakimeriksson/work/contiki-ng/tools/serial-io'
../../tools/serial-io/serialdump -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0
open: No such file or directory
```

It is not supposed to bring all the compiler arguments to the clang compiler when compiling serial dump, but for some reason your environment seems to bring them... Did you change anything in the makefiles?

One option for you would be to go into the tools/serial-io folder and compile serialdume there. Then it will likely work.

yvshudrenko, 2025-03-25:
It did, amazing, thanks!

closed 2025-03-25 by yvshudrenko